technoblogy/tiny-i2c #2: Setting bus speed?

sheffieldnick, 2018-10-20:
How easy is it to change the I2C bus speed using this library to a non-standard speed, like 10kHz?

It is just a case of picking different DELAY_T2TWI and DELAY_T4TWI durations?

Thanks

DirtyEngineer, 2018-10-25:
There is a "define" statement in the [TinyI2CMaster.h](https://github.com/technoblogy/tiny-i2c/blob/master/TinyI2CMaster.h) file; I think you can adjust as needed.

```
// Defines
#define TWI_FAST_MODE

#ifdef TWI_FAST_MODE                 // TWI FAST mode timing limits. SCL = 100-400kHz
#define DELAY_T2TWI (_delay_us(2))   // >1.3us
#define DELAY_T4TWI (_delay_us(1))   // >0.6us
#else                                // TWI STANDARD mode timing limits. SCL <= 100kHz
#define DELAY_T2TWI (_delay_us(5))   // >4.7us
#define DELAY_T4TWI (_delay_us(4))   // >4.0us
#endif
```

sheffieldnick, 2018-10-25:
What to those constants refer to? The LOW period and HIGH period of SCL? and for speeds <= 400kHz should DELAY_T2TWI >= DELAY_T4TWI + 0.7 us ?

Would (roughly) 50kHz be set by:

#define DELAY_T2TWI (_delay_us(9))   // >8.7us
#define DELAY_T4TWI (_delay_us(8))   // >8.0us

Thanks

closed 2022-04-21 by technoblogy