technoblogy/tiny-i2c #8: Compile error

AlexGyver, 2019-08-09:
Hello! I have a compile error. I use ATtiny core by Spence Konde v1.2.5 (and tyr 1.2.0), board ATtiny25/45/85 and have an error:
'''
cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin
'''

technoblogy, 2019-08-09:
I've just checked and it compiles OK for me. What's "tyr 1.2.0"?

What version of the Arduino IDE are you using? What platform?

AlexGyver, 2019-08-09:
I am using Windows 10, Arduino IDE 1.8.9, ATtiny 1.2.5, choose ATtiny25/45/85 without touching any other settings

technoblogy, 2019-08-09:
Can you run other programs on the ATtiny85, such as Blink?

AlexGyver, 2019-08-09:
I tried to compile on other PC, so it was successfull! Sorry for bothering you

technoblogy, 2019-08-09:
No problem - glad it worked!

closed 2019-08-09 by technoblogy

AlexGyver, 2019-08-09:
so, I have attiny85 working with core. But there is problem with tiny graphics sketch: are you shure that attiny can read OLED GDDRAM? There is some note from datasheet:

> To read data from the GDDRAM, select HIGH for both the R/W# (WR#) pin and the D/C# pin for 6800-series parallel mode and select LOW for the E (RD#) pin and HIGH for the D/C# pin for 8080-series parallel mode. **No data read is provided in serial mode operation.**
> ...
> The serial interface mode is always in write mode.  

So I have a very strange operation of your tiny graphics program. Maybe I have a bad Chinese copy? Driver is SSD1306.

technoblogy, 2019-08-10:
Which tiny graphics sketch are you referring to? Can you give a link to it?

AlexGyver, 2019-08-10:
This one http://www.technoblogy.com/show?23OS
And example from library tiny i2c. I think it is the same sketch, but with tinyWire instead of Wire

technoblogy, 2019-08-10:
As it says at the start of the article, that will only work with I2C OLED displays based on the SH1106 driver chip, for the reason you have highlighted in the datasheet. Sorry it's not clear from the listing - I'll add a note to clarify that.

I have another article that will work with SSD1306 displays; see:

[Tiny Function Plotter](http://www.technoblogy.com/show?2CFT)

It should be relatively easy to change it to work with the TinyI2C Library - let me know how you get on.

AlexGyver, 2019-08-10:
I thought that ssd1306 is the same as ssh1106, because you will find ssd1306 by searching ssh1106 on AliExpress :) 
Yes thanks, I've seen all your oled related programs, and plotter works.. only for horizontal lines. So I have solved my problem by buffering 2x2 pixels block on 128x32 display, so it's resolution is 64x16 now, with 255 byte buffer :D thank you for your codes, really helpful. Will try to buy SSH for future experiments

technoblogy, 2019-08-10:
Yes, there's a lot of confusion on AliExpress! Thanks for your comments.