technoblogy/tiny-i2c #11: NACK lock after reading from slave

oomek, 2020-04-28:
I have 2 Attinys85 takling, the master with your lib and the slave with TinyWire. Writing to slave goes very well, but when I request data I get a NACK lock for all future r/w operations. Do you know by any chance what might be the problem?

[image]

technoblogy, 2020-04-29:
OK, I’ll see if I can figure something out.

David

+------------------------------------------------------------+
David Johnson-Davies, Human-Computer Interface Ltd
17 Signet Court, Swanns Road, Cambridge, CB5 8LA, England.

Tel: +44 1223 314934, Fax: +44 1223 462562
Email: david@interface.co.uk, Web: http://www.interface.co.uk/
+------------------------------------------------------------+

technoblogy, 2020-05-03:
Sorry for the delay in having a look at this, but I can't see anything obviously wrong.

When you're requesting data are you specifying the number of bytes you want to read with the second parameter of TinyI2C.start(), or are you specifying the last byte with TinyI2C.readLast()?

oomek, 2020-05-03:
I tried both way with the same result.