technoblogy/tiny-i2c #12: Decluttering the signal

oomek, 2020-04-29:
Is it possible to tidy up the signal a little bit? I've been comapring the output of your lib with what the mcu of my TV is outputting and I've noticed some abnormalities: the unneccessary SDA spikes and the delay betwen the ACK bit. Is it doable, or it's some hardware limitation?

[image]

technoblogy, 2020-05-03:
The ATtiny85 doesn't have a dedicated I2C peripheral, and my routines use the general-purpose USI peripheral to emulate I2C. It's possible the emulation isn't perfect. Is this causing you a problem? You could try adjusting the delays in the code; they are inserted by the macros DELAY_T2TWI and DELAY_T4TWI.

oomek, 2020-05-03:
I was experimenting with delays few days ago, but it did not have any effect on those spikes except making them wider.

technoblogy, 2020-05-03:
Are these problems stopping it working for you?

closed 2022-04-21 by technoblogy