technoblogy/tiny-i2c #13: Library compilation error

Kozma04, 2020-06-15:
Hello,

This library looks great and I decided to integrate it in my project. The problem is that upon setting the board to AtTiny85 and compiling, I get some errors from the library source files (see attached).
Please fix this issue, I also tried defining the pins and ports manually, but with no luck.
[logs.txt](https://github.com/technoblogy/tiny-i2c/files/4780177/logs.txt)

technoblogy, 2020-06-15:
What core are you using? You need to use Spence Konde's ATTiny Core for these routines to work.

technoblogy, 2020-06-15:
I've added a note to the ReadMe. Let me know if that fixes it.

Kozma04, 2020-06-15:
Now it works perfectly. Thank you very much man, you saved me a lot of time. I highly appreciate your help :)

closed 2020-06-15 by Kozma04