technoblogy/tiny-i2c #14: SCL release?

rscheff, 2021-12-14:
Hi,

Line 95 is supposed to "release" SCL (that is, make it an input pin, and validate that no other device pulls it down).

However, the code keeps this as an output (PORTB PIN is set, rather than unset)...

Unless I am mistaken, this should be a 
PORT_USI_CL &= ~(1<<PIN_USI_SCL);

What am I missing?

https://github.com/technoblogy/tiny-i2c/blob/1029f6fca7c297eb887b601e3b0e352f71e60d1c/TinyI2CMaster.cpp#L95

technoblogy, 2021-12-14:
Thanks - I'll have a look.

rscheff, 2021-12-14:
Should have written

/* Release SCL to ensure that (repeated) Start can be performed */
	DDR_USI_CL &= ~(1<<PIN_USI_SCL);              // Release SCL.
	while (!(PIN_USI_CL & 1<<PIN_USI_SCL));     // Verify that SCL becomes high.
	DDR_USI_CL |= (1<<PIN_USI_SCL);
	
(make SCL an input, validate that the pull up still keeps SCL high, and make it a driven output again).

technoblogy, 2022-02-16:
Thanks for your comment. From the datasheet:

> The Serial Data (SDA) and the Serial Clock (SCL) pins are bi- directional and use open-collector output drives. The output drivers are enabled by setting the corresponding bit for SDA and SCL in the DDR Register.

As I understand it, because they are open-collector, setting SCL low leaves the slave free to pull it high, so it's OK the way it's written.

Was this a theoretical point, or have you encountered problems with the library as it's written?