technoblogy/tiny-i2c #15: Timing issues with a SSD1306 OLED

Karl255, 2023-02-21:
I've been having some issues with this library when using it with an SSD1306 OLED screen where the communication would completely break, usually only in the first second or two since power up. Although I managed to fix that by editing the delay times in TinyI2CMaster.cpp (lines 24 and 25):

```cpp
#define DELAY_T2TWI (_delay_us(1.3))   // >1.3us
#define DELAY_T4TWI (_delay_us(0.6))   // >0.6us
```

This works because the `_delay_us()` function takes a double and converts it to a cycle-accurate delay. The floating point math gets optimized away by the compiler.

After changing those lines everything worked flawlessly. I guess the SSD1306 is sensitive to low communication speeds. I didn't try this for low speed mode.

technoblogy, 2023-02-21:
Thanks - I'll update it after checking it a bit. What platform were you using it on, and what processor clock rate?

Karl255, 2023-02-21:
I used it on an ATtiny45 at 8 MHz.