technoblogy/tiny-i2c #18: ATiny406 compatibility

laserranger3000, 2024-06-30:
I'm trying to use this library to communicate with a MAX30102 sensor with an ATiny406.
There is a project that implements just that on an ATiny85: https://github.com/jeffmer/tinyPulsePPG

However, I can't get it to run on my microcontroller. As soon as the MAX30102 driver runs TinyI2C.start, the whole program freezes.

It's not a hardware problem, since using the Arduino wire library works just fine. I just can't use it due to the high amount of RAM it requires.

Is there any reason why it wouldn't run on an 406?

technoblogy, 2024-06-30:
It should certainly work with the ATtiny406. Have you rewritten the I2C commands to suit the TinyI2C library? The way the commands work is a bit different. I assume you've called `TinyI2C.init()`?

laserranger3000, 2024-06-30:
Yes, I've called the init function.

I finally got it to work by adding 5ms delay after the init function:
`//MAX30102
  TinyI2C.init();
  _delay_ms(5);
  sensor.begin();
  sensor.setup();`

Also, the MAX30102 library had one delay() call that I have missed, by changing that to _delay_ms() it works flawlessly.
I'm not sure why the 5ms delay is required. It was trial&error.

technoblogy, 2024-06-30:
Glad you got it working!

closed 2024-06-30 by technoblogy